trinamic/TMCL-IDE-Nightly #1: Linux support

s-light, 2018-03-26:
you say there is linux support in multiple places on your website and also the [`HowToStartLinuxTMCL-IDE.pdf`](https://github.com/trinamic/TMCL-IDE-Nightly/blob/master/HowToStartLinuxTMCL-IDE.pdf)  speaks about this-
but i did not find a download that seems too have the binaries or source for linux available.

quote from [your download site](https://www.trinamic.com/support/software/tmcl-ide/):
> TMCL-IDE nightly build and Linux version
> Visit our GitHub for nighly snapshots or latest Linux version of the TMCL-IDE.

and
> Latest Changes
> Version 3.0.20.0:
> 
>     Add TMCM1315, TMCM1370, TMCM3230 and TMCM3351
>     Add TMCM1111 and TMC8460-EVAL
>     Add TMC2041, TMC4661A Eval
>     Register browser with classic export menu
>     Show register values in info graphs 
>     Add LINUX version
>     Windows XP and older versions are not supported

---
i have looked into the history of the repository - and found
- https://github.com/trinamic/TMCL-IDE-Nightly/commit/284b12efeec88ed0601d07be4fb4004501f48120  `TMCL-IDE-3.0.19.0008.tar.gz` added
- https://github.com/trinamic/TMCL-IDE-Nightly/commit/30be5523f6bae6abba31368ee2ad3e398dbc5eb1 `TMCL-IDE-3.0.19.0008.tar.gz` changed
- https://github.com/trinamic/TMCL-IDE-Nightly/commit/e29a860a4aff5cae01263b3541e09e9724c6d31f `HowToStartLinuxTMCL-IDE.pdf` added
- https://github.com/trinamic/TMCL-IDE-Nightly/commit/917fea8e04bef20acb80d0e2f3561bf416d94bb1 ` TMCL-IDE-3.0.19.0008.tar.gz` deleted.

PS: would be really cool if you can use this repro for the actual software development and not only the (nightly)releases..

hadack, 2018-09-27:
Yes, that would be great.
Is even the latest version 3.0.22 on Linux available for download somewhere?

trinamic-mj, 2018-09-27:
Now, we are looking for an automatic method to deliver the binaries, includes all libraries needed. This version will be placed on our website first. After that automatically nightly builds could be possible.
This will comming soon

closed 2018-09-27 by trinamic-mj