trinamic/TMCL-IDE-Nightly #3: Could the user customized CPU board take advantage of TMCL-IDE?

huskier, 2019-03-09:
We've designed a motion control board, which integrated more than ten TMC4671s with a CPU included. Could our board take advantage of TMCL-IDE for PID tuning and parameters setting?

trinamic-mj, 2019-03-13:
In general not. The IDE is designed for Trinamic modules. You can use 4671-EVAL-Kit for optimizing an application.
But special for TMC4671, we designed RTMI-Interface (see trinamic.com) to communicate between IDE and TMC4671.

huskier, 2019-03-13:
Yes, we do have RTMI-interface at hand. However, the module is only used for one 4671 at a moment, and if we want to tuning another 4671, we have to take the connector and put into another 4671. It is quite inconvenient for multiple 4671s integrated on a big PCB board. So, is it possible to open the protocol?

trinamic-mj, 2019-03-13:
The protocol doesn't contain any address entry. So the IDE will address only the individual RTMI-Interfaces.

huskier, 2019-03-13:
Maybe this can be addressed by the DBGSPI_nSCS pin on the 4671. So we should design an adapter board between the RTMI-Interface and our customized board with 4671 chips. This may solve the problem.

closed 2019-03-14 by trinamic-mj