trinamic/TMCL-IDE-Nightly #11: Where is TMCLParam.tmc ?

dzid26, 2020-01-02:
Firmware manuals mention **TMCLParam.tmc** is provided with IDE, but I cannot find it in TMCL-IDE folder. 
It would be best if it was natively included in the editor.

[image]

alpytron, 2020-11-11:
Any update on this?
I could not find that file either.

trinamic-mj, 2020-11-11:
Ja, hab ich. Wollt auch grad ne Nightly bauen, komm aber nicht remote auf meinen Rechner.

Martin

EXTERNAL EMAIL

trinamic-ed, 2020-11-11:
Wenn der aus ist, dan ist Marius / Hagen oder eventuel Joshua noch in der Firma und könnte bestimmt einschalten.

EXTERNAL EMAIL

alpytron, 2020-11-11:
Is it possible to just simply upload that file somewhere here in github? :)

trinamic-mj, 2020-11-11:
Done

alpytron, 2020-11-11:
Thank you, Martin.
This is a huge help for us.