trinamic/TMCL-IDE-Nightly #17: Include file not found - TMCLParam.tmc

pettaa123, 2021-03-10:
ideLib: 1.2.0.0 - Qt 5.12.10
Version 0.0 64-Bit

On Windows including TMCLParam.tmc does work. On Ubuntu 20.04.2 LTS I get an error: Include file not found in line 1 using:
#include TMCLParam.tmc
I have one TMCLParam.tmc next to my program and one in \home\user\.config\TRINAMIC Motion Control GmbH & Co. KG\TMCL-IDE
Where does TMCL-IDE look for it?

trinamic-mj, 2021-03-10:
We are working on that issue. Problem is, file names are internally converted to uppercase letters. Renaming to TMCLPARAM.TMC could work?

pettaa123, 2021-06-30:
I renamed the file next to executable and the one in \home\user.config\TRINAMIC Motion Control GmbH & Co. KG\TMCL-IDE but still include file not found error. Where does the linux version search for the file?

trinamic-mj, 2022-01-25:
Upper and lower case is now taken into account.

closed 2022-01-25 by trinamic-mj