trinamic/TMCL-IDE-Nightly #18: Could you please open the ssoMap(Slave selection output) variable for the FT4222_SPIMaster_Init(...) function?

huskier, 2021-03-11:
The RTMI's FT4222H chip could connect to four SPI slaves with ss0o, ss1o, ss2o, and ss3o chip selection pins. If make the TMCL-IDE recognize the four SPI slaves, there should be a variable for the slave selection. We plan to integrate the RTMI module into our customized board for four TMC4671's tuning. Could you please open the ssoMap(Slave selection output) variable for the FT4222_SPIMaster_Init(...) function?

[image]

huskier, 2021-03-24:
@trinamic-mj  Hi, could you help us for the above issue?

trinamic-mj, 2021-03-24:
What I was told:
I think he is interested in how the IC is wired or configured. The IC is not used as master, as he imagines, but as slave. The TMC4671 is the SPI master, because it sends out the configured data in the PWM clock, in addition some IOs are also used for handshaking:

 ftd2xx_openEx((PVOID)devInfo.LocId, FT_OPEN_BY_LOCATION, &handlePortA)

ft4222_SPISlave_InitEx(handlePortA, SPI_SLAVE_NO_PROTOCOL)

I hope that can help you

huskier, 2021-04-05:
It is out of our expectation that the FT4222H chip is used as a SPI slave. It is still an issue to tune multiple TMC4671 with the USB-2-RTMI module or some other modules.

trinamic-ed, 2021-04-06:
Hi, 

our intention was never to use the RTMI interface to tune multiple TMC4671 with one interface. Our intention during the 4671 development was to have a really fast interface that can measure all the data synced with the PWM/Torque loop. This allows to measure 4 channels up to 100kHz synced with the PWM/Torque loop. During development, we saw that doing this with the FT4222H as spi master and the 4671 as the spi slave was far from feasible.

Regards,
     Enrico

huskier, 2021-04-06:
Thank you very much for the explanation, Enrico. @trinamic-ed

closed 2021-04-06 by huskier

marlesus, 2022-02-04:
Hello,

Is it possible to obtain a library (DLL or something) with which one can access the TMC4671's RTMI interface from python or other software?
I could see that there is that DBGSPI_TRG signal that makes this a customized interface (don't know why!?), rather than the standard SPI interface, and one cannot communicate through it with the standard FT4222H library...
What we would desire is to monitor the TMC4671 via the USB-2-RTMI interface without the need of the TMCL software.
Is this possible somehow ?

Thank you for your time!

Best regards,
Paul

marlesus, 2022-02-07:
> Hello,
>
> Is it possible to obtain a library (DLL or something) with which one can access the TMC4671's RTMI interface from python or other software? I could see that there is that DBGSPI_TRG signal [...]

@trinamic-ed , @trinamic-mj , @trinamic-LH : Can one of you please answer to my request above ? We are really interested in the USB-2-RTMI interface, if we could access it somehow from outside TMCL-IDE...

trinamic-ed, 2022-02-08:
Hi Paul,

we are using the RTMI in the TMCL-IDE with the standard FT4222H library:

[FT4222HCommandHandler.cpp.txt](https://github.com/trinamic/TMCL-IDE-Nightly/files/8021467/FT4222HCommandHandler.cpp.txt)

In the moment we have no python based access for the RTMI interface. We are working on a major update on pytrinamic. Afterwards RTMI support would be a good fit for pytrinamic as well. Until then, I can support with some code samples on how to use the RTMI and how to configure it for measurements. So if there are more questions about details, I will try to answer.

Regards,
    Enrico

marlesus, 2022-02-08:
Hi Enrico @trinamic-ed ,

Thank you very much for the reply and info provided!
Yes, it would be great if you could send me some code samples on how to use the RTMI!
I am using the pytrinamic already.

Best regards,
Paul

marlesus, 2022-03-06:
Hi Enrico,

Could you please attach also the header files included in the .cpp file above?
Some key constants are missing (e.g. SYNCWORD, GPIO_DATA_AVAILABLE etc.), and I believe they are defined in:
#include "libTMCComm/spi/FT4222H/FT4222HCommandHandler.h"
#include "libTMCComm/spi/FT4222H/FT4222HDataThread.h"

Thank you!

Best regards,
Paul

trinamic, 2022-03-07:
Hi,

here is some more code to access the RTMI wit Qt:

[FT4222H_Example.zip](https://github.com/trinamic/TMCL-IDE-Nightly/files/8196532/FT4222H_Example.zip)

Do you have a first example to access the FT4222H library with Python?

Regards,
     Enrico

marlesus, 2022-03-07:
Hi Enrico, @trinamic-ed 

Thank you very much for the data provided!
About the Python example, not quite. It would be great if you could provide me something already containing the handshaking with the "DBG_TRIGGER" signal, or whatever you have in this direction...

Thank you again!

Best regards,
Paul