trinamic/TMCL-IDE-Nightly #19: Where does the monitoring motor supply voltage displayed on the left-top  corner of TMCL-IDE come from?

huskier, 2021-03-16:
Hi, we wonder where the monitoring motor supply voltage displayed on the left-top  corner of TMCL-IDE comes from? TMC4671's AGPI_A, AGPI_B, or ADC_VM?

We guess it is from ADC_VM. However, although we have the same ADC_VM input curcuit design following the TMC4671-eval board, we get different displayed value (47V~52V) from the evaluation board(23.4V). The motor supply is the same, 24V. 

[image]

huskier, 2021-03-24:
@trinamic-mj  Hi, could you help us for the above issue?

trinamic-mj, 2021-03-24:
Hi, This is not the right forum for this kind of help. Please use the Technical Support from trinamic.com.
someone told me it was iirc ADC_VM

huskier, 2021-03-27:
Now, we have known where the displayed motor supply voltage in the TMCL-IDE  is from. It is from the L-Board's VM Measurement as the following diagram. Before, we always think it is from the TMC4671's ADC_VM.

Thanks for your help! @trinamic-mj 

[image]

closed 2021-03-27 by huskier