trinamic/TMCL-IDE-Nightly #22: TMCL-IDE doesn't work on Ubuntu 20.04: "qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in """

Chris-Idema-WizzDev, 2021-11-09:
If I run the linux version I get these errors:
```
./TMCL-IDE.sh
	linux-vdso.so.1 (0x00007ffcce1c6000)
	libIDECore5.12.11.so.1 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libIDECore5.12.11.so.1 (0x00007fa2d2aea000)
	libIDEComm5.12.11.so.1 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libIDEComm5.12.11.so.1 (0x00007fa2d287a000)
	libIDEGui5.12.11.so.1 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libIDEGui5.12.11.so.1 (0x00007fa2d2469000)
	libIDEPlugin5.12.11.so.1 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libIDEPlugin5.12.11.so.1 (0x00007fa2d20f0000)
	libIDEDbg5.12.11.so.1 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libIDEDbg5.12.11.so.1 (0x00007fa2d1ebf000)
	libIDECommon5.12.11.so.1 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libIDECommon5.12.11.so.1 (0x00007fa2d1c87000)
	libQt5Svg.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5Svg.so.5 (0x00007fa2d1a30000)
	libQt5PrintSupport.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5PrintSupport.so.5 (0x00007fa2d17be000)
	libQt5Charts.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5Charts.so.5 (0x00007fa2d140a000)
	libQt5Widgets.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5Widgets.so.5 (0x00007fa2d0bba000)
	libQt5Gui.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5Gui.so.5 (0x00007fa2d036a000)
	libQt5Network.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5Network.so.5 (0x00007fa2cffc2000)
	libQt5SerialPort.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5SerialPort.so.5 (0x00007fa2cfda9000)
	libQt5SerialBus.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5SerialBus.so.5 (0x00007fa2cfb66000)
	libQt5Xml.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5Xml.so.5 (0x00007fa2cf92d000)
	libQt5Core.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5Core.so.5 (0x00007fa2cf18e000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007fa2cf0e8000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fa2cf0c5000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007fa2ceee3000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007fa2ced94000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007fa2ced79000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fa2ceb85000)
	libQt5OpenGL.so.5 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libQt5OpenGL.so.5 (0x00007fa2ce92e000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fa2ce912000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007fa2ce90c000)
	libicui18n.so.56 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libicui18n.so.56 (0x00007fa2ce471000)
	libicuuc.so.56 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libicuuc.so.56 (0x00007fa2ce0b9000)
	libicudata.so.56 => /home/username/Downloads/TMCL-IDE-3.5.0.0/5.12.11/./bin/libicudata.so.56 (0x00007fa2cc6d6000)
	libgthread-2.0.so.0 => /lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007fa2cc6d1000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007fa2cc5a8000)
	/lib64/ld-linux-x86-64.so.2 (0x00007fa2d32e0000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007fa2cc4ee000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007fa2cc4ba000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007fa2cc447000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007fa2cc30a000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007fa2cc2e0000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007fa2cc2d8000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007fa2cc2d0000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007fa2cc2b6000)
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

./TMCL-IDE.sh: line 16: 41475 Aborted                 (core dumped) $dirname/$appname "$@"
```
I also tried to run both windows versions (install and non-install) with Wine. Both run, but when I try accessing serial port it freezes (I linked COM1 to /dev/ttyUSB0). Our client created a windows program to control a motor via Trinamic protocol, this executable does run via Wine and does interact with the serial port. So it is not a hardware problem. On a separate windows PC it works too, but I develop on linux so that is not an option for me.

trinamic-mj, 2021-11-09:
I mean the Qt-Lib. we use 5.12.11 for IDE

Chris-Idema-WizzDev, 2021-11-09:
```
qmake --version

QMake version 3.1
Using Qt version 5.12.8 in /usr/lib/x86_64-linux-gnu
```
So I assume it is 5.12.8

trinamic-mj, 2021-11-09:
I added platforms.tar.gz here. Unpack platforms folder to the TMCL-IDE folder.

closed 2021-11-12 by trinamic-mj

Chris-Idema-WizzDev, 2021-11-18:
Thank you. It works.