trinamic/TMCL-IDE-Nightly #23: Linux version of IDE will not run on Ubuntu 20.04 LTS

feylen, 2021-11-30:
I have worked for about 4 hours to try and get the Linux version of the TMCL-IDE to run with no luck. No matter the QT5 version, dev packages and support on the internet of similar issues, I can not get the IDE to launch. Any help VERY WELCOME. Please let me know any additional information that you might need to diagnose the issue.

# uname -a (officially ZorinOS 16 Pro)
Linux 5.11.0-41-generic #45~20.04.1-Ubuntu SMP Wed Nov 10 10:20:10 UTC 2021 x86_64 GNU/Linux

(base) 5.12.11>./TMCL-IDE.sh 
LD_LIBRARY_PATH: /home/seanorear/Applications/5.12.11/./bin
	linux-vdso.so.1 (0x00007ffef33ec000)
	libIDECore5.12.11.so.1 => /home/seanorear/Applications/5.12.11/./bin/libIDECore5.12.11.so.1 (0x00007f367c9ce000)
	libIDEComm5.12.11.so.1 => /home/seanorear/Applications/5.12.11/./bin/libIDEComm5.12.11.so.1 (0x00007f367c75e000)
	libIDEGui5.12.11.so.1 => /home/seanorear/Applications/5.12.11/./bin/libIDEGui5.12.11.so.1 (0x00007f367c34d000)
	libIDEPlugin5.12.11.so.1 => /home/seanorear/Applications/5.12.11/./bin/libIDEPlugin5.12.11.so.1 (0x00007f367bfd4000)
	libIDEDbg5.12.11.so.1 => /home/seanorear/Applications/5.12.11/./bin/libIDEDbg5.12.11.so.1 (0x00007f367bda3000)
	libIDECommon5.12.11.so.1 => /home/seanorear/Applications/5.12.11/./bin/libIDECommon5.12.11.so.1 (0x00007f367bb6b000)
	libQt5Svg.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5Svg.so.5 (0x00007f367b914000)
	libQt5PrintSupport.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5PrintSupport.so.5 (0x00007f367b6a2000)
	libQt5Charts.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5Charts.so.5 (0x00007f367b2ee000)
	libQt5Widgets.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5Widgets.so.5 (0x00007f367aa9e000)
	libQt5Gui.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5Gui.so.5 (0x00007f367a250000)
	libQt5Network.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5Network.so.5 (0x00007f3679ea8000)
	libQt5SerialPort.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5SerialPort.so.5 (0x00007f3679c8d000)
	libQt5SerialBus.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5SerialBus.so.5 (0x00007f3679a4a000)
	libQt5Xml.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5Xml.so.5 (0x00007f3679811000)
	libQt5Core.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5Core.so.5 (0x00007f3679074000)
	libGL.so.1 => /lib/x86_64-linux-gnu/libGL.so.1 (0x00007f3678fd1000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f3678fae000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f3678dca000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f3678c7b000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f3678c60000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f3678a6e000)
	libQt5OpenGL.so.5 => /home/seanorear/Applications/5.12.11/./bin/libQt5OpenGL.so.5 (0x00007f3678817000)
	libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f36787f9000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f36787f3000)
	libicui18n.so.56 => /home/seanorear/Applications/5.12.11/./bin/libicui18n.so.56 (0x00007f367835a000)
	libicuuc.so.56 => /home/seanorear/Applications/5.12.11/./bin/libicuuc.so.56 (0x00007f3677fa2000)
	libicudata.so.56 => /home/seanorear/Applications/5.12.11/./bin/libicudata.so.56 (0x00007f36765bf000)
	libgthread-2.0.so.0 => /lib/x86_64-linux-gnu/libgthread-2.0.so.0 (0x00007f36765b8000)
	libglib-2.0.so.0 => /lib/x86_64-linux-gnu/libglib-2.0.so.0 (0x00007f367648f000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f367d1c4000)
	libGLdispatch.so.0 => /lib/x86_64-linux-gnu/libGLdispatch.so.0 (0x00007f36763d7000)
	libGLX.so.0 => /lib/x86_64-linux-gnu/libGLX.so.0 (0x00007f36763a3000)
	libpcre.so.3 => /lib/x86_64-linux-gnu/libpcre.so.3 (0x00007f3676330000)
	libX11.so.6 => /lib/x86_64-linux-gnu/libX11.so.6 (0x00007f36761f1000)
	libxcb.so.1 => /lib/x86_64-linux-gnu/libxcb.so.1 (0x00007f36761c7000)
	libXau.so.6 => /lib/x86_64-linux-gnu/libXau.so.6 (0x00007f36761c1000)
	libXdmcp.so.6 => /lib/x86_64-linux-gnu/libXdmcp.so.6 (0x00007f36761b9000)
	libbsd.so.0 => /lib/x86_64-linux-gnu/libbsd.so.0 (0x00007f367619f000)
qt.qpa.plugin: Could not find the Qt platform plugin "xcb" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

./TMCL-IDE.sh: line 16: 40557 Aborted                 $dirname/$appname "$@"

trinamic-mj, 2021-12-01:
We added platforms.tar.gz here. Please unpack platforms folder to the TMCL-IDE folder.

feylen, 2021-12-01:
> {
> "target": "github.com/trinamic/TMCL-IDE-Nightly/issues/23#issuecomment-983419535",
> "url": "github.com/trinamic/TMCL-IDE-Nightly/issues/23#issuecomment-983419535",
> "name": "View Issue"
> },
> "description": "View this Issue on GitHub",
> "publisher": {
> "name": "GitHub",
> "url": "github.com"
> }
> }
> ]

That did it. Thank you.Sean O'Rear

closed 2021-12-08 by trinamic-mj