trinamic/TMCL-IDE-Nightly #25: TMCL-IDE 3.6.0.0 (Windows) has a sluggish UI

cradnovichnar, 2022-06-11:
I have been using this version for a few days and the UI freezes and becomes unresponsive multiple times. I have closed all programs and tried different USB ports (including direct connection) and I still get freezes. The previous version (not sure, but thinking 3.5.0.0) seemed to be more sluggish than this version. Back in September 2021, I used what ever version was out then and didn't notice this sluggishness.

neliooliveira, 2022-06-12:
With 3.6 I can't even start the wizard, the program crashes every time.

Using 3.5 works without any problem.

Also tried nightly 3.5.2 and also crashes.

cradnovichnar, 2022-06-12:
I was able to get most of the features working with 3.6.0.0 after installing the windows driver. Torque, velocity, and position mode UI widgets don't seem to set the appropriate variables, so I have had to use the register browser. It also seems like the plots in the wizard are using a low sampling frequency because the sine waves are triangular. I don't remember this in the previous build which I think was 3.5.0.0 but I will confirm this again.

trinamic-mj, 2022-06-13:
The chrash will be fixed asap.

closed 2022-06-14 by trinamic-mj