trinamic/TMCL-IDE-Nightly #26: TMCL-IDE 3.6.x (Windows 10) is almost unusable

huskier, 2022-06-30:
Hi, after installing TMCL-IDE 3.6.x(Windows 10), the software is always crashes when we select tuning tools, barely successful for tuning.

The latest version of 3.6.2 (nightly build) is the same as the stable version 3.6.0, barely usable. 

Thanks for your investigation.

Best regards,
Youbao

trinamic-mj, 2022-06-30:
Hi,
can add module with firmeware version and the interface?

huskier, 2022-07-01:
Hi, thanks for response. We use USB-2-RTMI module for TMC4671 tuning.

trinamic-mj, 2022-07-01:
Hi, I can open all the tools without crashing.

The only thing is that the wizard got a new axis parameter, where axis parameters with RTMI interface are not allowed. A lot of red windows appear, which should be harmless(I'll fix that).

huskier, 2022-07-02:
Thanks. We're looking forward to the new stable version.

trinamic-mj, 2022-07-04:
Not that we misunderstand each other:
This version is stable for me. And the red windows are only unnecessary, and do not cause a crash.
Can you describe in more detail when something doesn't work?

trinamic-mj, 2022-07-12:
Do you have more specifics on when crashes occur?

huskier, 2022-07-13:
Sorry for the delay reply.

The most time when we click the tuning parts, the GUI is unresponding, and the mouse just become circle, and always circling, no stop. We have to kill the program with task manager.

The possibility for crashing is more than work properly. The current version(3.6.0 and 3.6.2) is almost unusable for us. 

We do not know whether the TMCL-IDE issue is related to the OS version. Our windows version is "Windows 10 Enterprise LTSC".

trinamic-mj, 2022-07-14:
I have asked around, but the behavior is not known.
- is the 4671 powered
- firmware one of the latest
- did older IDE versions work
- before you use a tool, the other tools work e.g. Torque Graph

huskier, 2022-07-15:
The issues is really presented here in our laboratory.
- The TMC4671 is definitely powered. 
- We use USB-2-RTMI module(the 1.0 version, there is no isolation chips between the FT4222HQ and the TMC4671's SPI signals), so the firmware does not matter. We do not use the L-Board for tuning.
- Yes , the older IDE(3.0.24 ) works properly. The version 3.2.x is workable. But, 3.5.0 is the same as 3.6.x, not work as expected. We do not try other versions. We are now using the V3.0.24 for tuning.
- No, the AXIS related part is not worked properly, it's always running into unresponding state.

EthChil, 2022-08-06:
I think this might be gaslighting?

EthChil, 2022-08-06:
I am also running into a similar issue on Windows 11. We are using the TMC4671+6100-BOB and the program is quite unresponsive the mouse disappears at times make it extremely difficult to use.

clayhaight01, 2022-08-07:
Also running with TMC 4671 and RTMI, but on Windows 10. This software is a disaster. The main builds here: https://www.trinamic.com/support/software/tmcl-ide/ don't even connect to the chip, but for some reason the nightly build does. Despite the nightly build being able to connect, it is pretty much useless and a complete waste of time. Opening the wizard menu and weasel configurator crashes the software 50% of the time, and brings up 10 black/red error boxes every time it is opened. Menus are slow/unresponsive, mouse disappears frequently, data rate on the graphs in the wizard are extremely slow and barely resemble sine waves when they should.

I genuinely think it would've been faster to write directly to registers without the TMCL IDE. Horrible waste of time, please fix it!

trinamic-mj, 2022-08-08:
The Notification error was fixed in latest nightly

closed 2022-08-08 by trinamic-mj