trinamic/TMCL-IDE-Nightly #27: Must have

sdrahtpt, 2022-07-13:
Dear Trinamic,
I really love your work, but i just use the TMC8461 as a Data Bridge. I write my process Data to the RAM and read my process Data from the TMC that is written from the EtherCat Master to the ram or the TMC.
But i need some documentation, what to write to get the TMC in operational mode.
Just the must have commands to get the TMC8461 running. And to stay in operational mode.
My SPI is working perfect, but i just know the commands:
#define TMC846x_AL_STATE_OVERRIDE 0x02C0
#define TMC8461_ESC_AL_CONTROL          0x0120
#define TMC8461_ESC_AL_STATUS           0x0130
#define TMC8461_ESC_AL_CODE             0x0134
#define TMC8461_ESC_PDI_CTRL            0x0140

So do i really just have to write 

TMC8461_ESC_AL_CODE=0x0000
TMC8461_ESC_AL_EVENT_MASK_LO = 0xFF0E
TMC8461_ESC_AL_CONTROL,=8

I can read a 5 from TMC8461_ADR_INTERFACE_TYPE

So the SPI is working.
But my SPS tells me that the Trinamic TMC8461 never cahnges to Operational Mode.
it is just going from init to preop and then back to init:
[image]

Or is there a trigger i have to send?

/Stefan

trinamic-mj, 2022-07-14:
I have asked around:
Do you have also externally on the IC the pin "DEVICE EMULATION" correctly set?
This switches the IC into the mode that the state machine is operated manually and not via a slave stack / external MCU.

sdrahtpt, 2022-07-14:
Yes, i put the jumper to "Device Emulation".
The "Run" LED is blinking.
If i put the Jumper not ot "device Emulation" the Run LED is off.
So there might be some commands to put the TMC8461 to preop mode.
My twincat System tells me that the AL Status is 0x0012. It should be 0x0002.
[image]
If you have a running code snipped with the content that i have to send to communicate, it would be fine to get.
With "Landungsbrücke" the problem is solved.->After Firmwareupdate and the usage of TMCL-IDE.
So "Landungsbrücke" must have some commands. I do not like to put my sniffer to the SPI. The SPI is extremly "active".
There should be a easier way to solve that issue:-)
 /Stefan

trinamic-mj, 2022-07-14:
Yes, you have to set "Device Emulation" in Eeprom

sdrahtpt, 2022-07-14:
Now i have it:
It is a bit more complicate: I really checked some things.

1. TMC8461 in EMULATION MODE. "Landungsbrücke" attached.
2. TMC8461 not in EMULATION MODE. "Landungsbrücke" attached.
3. TMC8461 in EMULATION MODE.  "Landungsbrücke" not attached.
4. TMC8461 not in EMULATION MODE.  "Landungsbrücke" not attached.
Just one combination is working propperly.  TMC8461 not in EMULATION MODE. "Landungsbrücke" attached.
So, i need the triggering and init of the "Landungsbrücke".
Do you have some code for me?
/Stefan

trinamic-mj, 2022-07-15:
I think the TwinCATs Eeprom Ediitor can do this:
[image]

closed 2022-08-08 by trinamic-mj