trinamic/TMCL-IDE-Nightly #29: TMCL-IDE 3.6.x (Windows 10) is (still) almost unusable

clayhaight01, 2022-08-08:
The previous thread titled "TMCL-IDE 3.6.x (Windows 10) is almost unusable" was closed because apparently the issues were fixed in the new nightly build (3.6.4.0). The issues were not fixed, al the issues still persist. See image below:
[image]

trinamic-mj, 2022-08-08:
I just compile a new version.

trinamic-mj, 2022-08-08:
I have simply replaced the zip

clayhaight01, 2022-08-09:
Every new version gets slower and slower. This recent version takes around 1sec/line of TMC code to assemble.

trinamic-mj, 2022-08-09:
[image]
Have you made any changes here?

huskier, 2022-08-11:
@trinamic-mj Hi, the latest version V3.6.4 is still unusable. It always says "No response", and the GUI just freeze there!

Our system is Win10_x64. Is it possible that the released version is x86, not x64 which cause the problem? Or, are there any language settings or local settings which cause the issue? 

BTW, is it possible to open source part of the TMCL-IDE? Keep your products related code in private. In this way, we could be possible to help a little bit.

[image]

trinamic-mj, 2022-08-11:
I do not know now how I can help further. Everything seems to work: 
[image]

huskier, 2022-08-11:
It's very weird.  We've really encountered the "No response" stuck.

Our OS info as following:
Windows 10 Enterprise LTSC
Version 1809, Internal version 17763.3046
x64

We've found the program is 32bit. Could you please compile a x64bit release version?
[image]

Or, maybe we try other Win10 OS.

huskier, 2022-08-11:
Actually, V3.2.0 is OK. The behavior of V3.6.4 is different from V3.2.0, and it is very sluggish.

@trinamic-mj, do you really tuning something, or just open some windows?

trinamic-mj, 2022-08-11:
Yes, I tuned some values with the Wizard

huskier, 2022-08-11:
The Wizard part is probably no problem. We think the issue mostly comes from the tuning part. Maybe the GUI update frequency is the main problem. Sometimes, it is needed to wait several seconds to input a number(or parameter) ; sometimes, the mouse becomes a waiting circle, just "No Response".

There are smooth version there, say V3.2.0, and the previous other versions before V3.2.0. Maybe there are some important variables which have been changed during the upgrading development.

We mainly use the IDE to tune TMC4671 through RTMI interface.

closed 2022-08-22 by trinamic-mj