trinamic/TMCL-IDE-Nightly #32: Generated TMCL/PC code for Landungsbruecke small  after weasel wizard gives "Connection to module not open" error

joostm8, 2023-01-27:
Hi all, I have a TMC4671 Eval kit that I'm trying to get up and running. Unfortunately, I get an error when running the generated code from the Weasel wizard:

`TMCL/PC runtime error in line 9:
Connection to module not open.`

 Steps to reproduce the error are as follows:

1. Run Weasel configuration wizard for TMC4671
2. Export script to .tpc file
3. Open TMCL/PC Host
4. Open script
5. Run script

Note that the motor runs properly during and after the wizard, e.g. I can go and tune the PI controllers without problem. I need the working script though since there is no persistent memory on the chip.

I don't know if this is specifically related to the IDE, I guess it's mainly the combination of Landungsbruecke Small and the IDE that is the root cause. The latest non-nightly release doesn't seem to support the landungsbruecke at all (shows up as ID1:UNKNOWN [V3.08]), and even in this nightly release I have to manually hit scan for the TMC4671-EVAL to show up as motion controller ic, while in the demos this process is seamless.

Hardware:

- Landungsbruecke (Small) V2.0
- TMC4671 EVAL V1.2
- TMC-UPS-2A24V-A-EVAL V1.1

Software:

- Landungsbruecke (Small) firmware: V3.08
- TMCL IDE Nightly 3.6.6.0

Code that yields the error:

```
// Select module
#module 1 COM6/USB/id1/Landungsbruecke (Small)

// Use TMC4671 register addresses
#include TMC4671_register_addresses.tpc
// (C:/Users/Joost/AppData/Roaming/TRINAMIC Motion Control GmbH & Co. KG/TMCL-IDE/TMCL-Script/TMC4671_register_addresses.tpc)

// Motor type &  PWM configuration
WMC MOTOR_TYPE_N_POLE_PAIRS, 0, $00030004, 1 
WMC PWM_POLARITIES,          0, $00000000, 1 
WMC PWM_MAXCNT,              0, $00000F9F, 1 
WMC PWM_BBM_H_BBM_L,         0, $00000A0A, 1 
WMC PWM_SV_CHOP,             0, $00000007, 1 

// ADC configuration
WMC ADC_I_SELECT,        0, $18000100, 1 
WMC dsADC_MCFG_B_MCFG_A, 0, $00100010, 1 
WMC dsADC_MCLK_A,        0, $20000000, 1 
WMC dsADC_MCLK_B,        0, $20000000, 1 
WMC dsADC_MDEC_B_MDEC_A, 0, $014E014E, 1 
WMC ADC_I0_SCALE_OFFSET, 0, $01008212, 1 
WMC ADC_I1_SCALE_OFFSET, 0, $0100816E, 1 

// Open loop settings
WMC OPENLOOP_MODE,            0, $00001000, 1 
WMC OPENLOOP_ACCELERATION,    0, $0000003C, 1 
WMC OPENLOOP_VELOCITY_TARGET, 0, $FFFFFFFB, 1 

// Feedback selection
WMC PHI_E_SELECTION, 0, $00000002, 1 
WMC UQ_UD_EXT, 0, $000007D6, 1 

// ===== Open loop test drive =====

// Switch to open loop velocity mode
WMC MODE_RAMP_MODE_MOTION, 0, $00000008, 1 

// Rotate right
WMC OPENLOOP_VELOCITY_TARGET, 0, $0000003C, 1 
WAIT TICKS, 1, 2000

// Rotate left
WMC OPENLOOP_VELOCITY_TARGET, 0, $FFFFFFC4, 1 
WAIT TICKS, 1, 4000

// Stop
WMC OPENLOOP_VELOCITY_TARGET, 0, $00000000, 1 
WAIT TICKS, 1, 2000
WMC UQ_UD_EXT, 0, $00000000, 1 
```

trinamic-mj, 2023-02-01:
#module 1 COM6/USB/id1/Landungsbruecke_**"Space"**_(Small)
maybe the space before (Small) causes the issue.
Could only a "COM6/USB/id1" help?

we will work on it

trinamic-mj, 2023-02-10:
V3.6.7.0