trinamic/TMCL-IDE-Nightly #35: Microstep Resolution not written to module when using "Module Settings"

Andy1978, 2023-05-12:
Both, the stable TMCL-IDE 3.6.0.0 and testing 3.6.8.1 both show the same problem:

[image]

If you use "Module Settings" -> "Microstep" Tab and use the dropdown to change the microstep resolution I would expect to see some "SAP 140, 0, X" command on the logfile but the changed value isn't send to the module.

"Store Current Tab" sends STAP 140, 141, 142 as expected.

I think this obviously is a bug in the IDE, perhaps the programmer forgot the "onChange callback action" for the DropDown and the other checkboxes on that tab.

Andy1978, 2023-05-19:
Same for StealthChop, SpreadCycle and CoolStep Tab. Only changes on the "Current" tab are written to the module, which makes the IDE quite useless for TMCM-1076.

trinamic-mj, 2023-05-23:
We will take care of it as soon as possible.

closed 2023-08-11 by trinamic-mj

Andy1978, 2023-08-12:
Okay, this has been closed after 3 months (as soon as possible) as completed. Thank you for your quick response

But I can't find a new download where this is fixed.

trinamic-mj, 2023-08-17:
Sorry for the delay, the Linux version should be available next week.

trinamic-mj, 2023-08-25:
Add v3.11.0