trinamic/TMCL-IDE-Nightly #39: No Register Browser in TMCL-IDE 3.11.0  on Ubuntu 20.04

idraney, 2023-09-01:
Hi @trinamic-mj , 

I'm a newb to Trinamic devices, but I think I stumbled on a bug in TMCL-IDE v3.11.0 on Ubuntu 20.04.  Being a newb, I couldn't find the Register Browser.  However, after installing TMCL-IDE v3.6.8.1, the Register Browser showed up just fine.  

I tried to copy the libtools_register_browser*.so files from v3.6.8.1 to v3.11.0, but the failure persists.

This is more of a heads up than a complaint.  I'm happy to use 3.6.8.1.

Thanks!
Ian

p.s. Warning messages from the output log when running V3.11.0/TMCL_IDE.sh below.

```
ian@9SE:~/TMCL_IDE/V3.11.0$ ./TMCL-IDE.sh
. . .
14:01:08:428> #8F672FC0(IDE-Main) WRN: default :Could not load plugin:  "/home/ian/TMCL_IDE/V3.11.0/Plugins/Connections/libcan_socket.so" QDateTime(2023-08-25 03:33:01.000 PDT Qt::LocalTime) "Failed to extract plugin meta data from '/home/ian/TMCL_IDE/V3.11.0/Plugins/Connections/libcan_socket.so'"
. . .
14:01:08:560> #8F672FC0(IDE-Main) WRN: default :Could not load plugin:  "/home/ian/TMCL_IDE/V3.11.0/Plugins/Vendor/Tools/libtools_register_browser12Bit.so" QDateTime(2023-08-25 03:35:02.000 PDT Qt::LocalTime) "Cannot load library /home/ian/TMCL_IDE/V3.11.0/Plugins/Vendor/Tools/libtools_register_browser12Bit.so: (/home/ian/TMCL_IDE/V3.11.0/Plugins/Vendor/Tools/libtools_register_browser12Bit.so: undefined symbol: _ZTV12ToolLineEdit)"
. . .
14:01:13:342> #8F672FC0(IDE-Main) WRN: main.xml "Tool 'CoolStep & StallGuard': Missing parameter 'SG2Threshold' for motor number 0"**
**14:01:13:342> #8F672FC0(IDE-Main) WRN: main.xml "Tool 'CoolStep & StallGuard': Missing parameter 'SG2FilterEnable' for motor number 0"**
**14:01:13:342> #8F672FC0(IDE-Main) WRN: main.xml "Tool 'CoolStep & StallGuard': Missing parameter 'SG4FilterEnable' for motor number 0"
. . . 
```

trinamic-mj, 2023-09-18:
I will check this ASAP

trinamic-mj, 2023-09-18:
Unfortunately, a builderror had crept in. The version is reloaded.

closed 2023-09-18 by trinamic-mj